gazebosim/gazebo_test_cases #1101: gz-common: Profiler

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/common/6/profiler.html)
* [Source](https://github.com/gazebosim/gz-common/tree/gz-common6/tutorials/profiler.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Sanjay1911, 2024-09-01:
I'm currently testing this and managed to install it after a couple of tries. 

The instruction at [https://gazebosim.org/api/common/6/profiler.html](GZ_Common) page is clear but in my opinion it's better to use the profiler.cc script which is already inside the /examples directory. This would mean `add_executable(profiler_example profiler.cc)` should be replaced to `add_executable(profiler_example examples/profiler.cc)` 

Also there was another error when running `cmake ..` which is specific to FreeImage and the error is as follows:

> CMake Error at graphics/src/CMakeLists.txt:5 (target_link_libraries):
  Target "my_gz_common6_project-graphics" links to:
> FreeImage::FreeImage
> but the target was not found.  Possible reasons include:
> * There is a typo in the target name.
> * A find_package call is missing for an IMPORTED target.
> * An ALIAS target is missing.

Another point to note is that ./profiler_example executable is located in the **/gz-common/build/bin** directory and if you run the ./profiler_example executable you will get a  
> bash: ./profiler_example: No such file or directory

For someone with less cmake experience trying to follow the tutorial and also for the above FreeImage target not found error, this is the [CMakeLists.txt](https://github.com/user-attachments/files/16829792/CMakeLists.txt) file for successful installation and testing.

Since I'm a Robotics student and not a professional Dev, I will wait for advice and proceed with a PR if needed.

Sanjay1911, 2024-09-01:
Successful installation screenshots for **Use the source path (Linux)** option

[image]
[Screencast from 2024-09-01 18-47-27.webm](https://github.com/user-attachments/assets/4ee1df7f-f2b2-47dc-b20a-f3e73c263a76)

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde

azeey, 2024-09-03:
Thanks for testing @Sanjay1911!!

> This would mean add_executable(profiler_example profiler.cc) should be replaced to add_executable(profiler_example examples/profiler.cc)

The instructions say to download `profiler.cc` and provides a link. If the user downloads the file as instructed places it in the same directory as the `CMakeLists.txt` file, your suggestion will not work. The example in https://github.com/gazebosim/gz-common/tree/gz-common6/examples also has the `CMakeLists.txt` and `profiler.cc` files in the same directory, so I think we should leave it as `add_executable(profiler_example profiler.cc)`

> Also there was another error when running cmake .. which is specific to FreeImage and the error is as follows:

This looks like you don't have freeimage installed. It should have been installed with `gz-common`. Can you describe how you installed gz-common?

> Another point to note is that ./profiler_example executable is located in the /gz-common/build/bin directory and if you run the ./profiler_example executable you will get a

hmm, following the example, the file was in the `build` directory for me, and since the instructions say to do `cd build`, running `./profiler_example` should work.