gazebosim/gazebo_test_cases #1104: gz-fuel-tools: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/fuel_tools)
* [Source](https://github.com/gazebosim/gz-fuel-tools/blob/gz-fuel-tools10/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-02:
Interested in !

avanmalleghem, 2024-09-03:
- typo " then used "
- remove List all models & Download a model (the helper is sufficient)
- remove dependencies section

Will create a PR soon

avanmalleghem, 2024-09-03:
PR opened : https://github.com/gazebosim/gz-fuel-tools/pull/443
There is a question in it : it looks like code section are invisible live. Maybe a space line improve it ? I don't know how to test it.

iche033, 2024-09-03:
> There is a question in it : it looks like code section are invisible live. Maybe a space line improve it ? I don't know how to test it.

i think the  ```{.sh} syntax does not work live. We'll probably need to change it to \`\`\`sh or \`\`\`bash

ahcorde, 2024-09-04:
> syntax

@iche033 https://github.com/gazebosim/gz-fuel-tools/pull/445

closed 2024-09-04 by ahcorde