gazebosim/gazebo_test_cases #1119: gz-sim: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/sim)
* [Source](https://github.com/gazebosim/gz-sim/blob/gz-sim9/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

avanmalleghem, 2024-09-01:
Code of Conduct and Contributing are not in the right order in the table of content. Will create a PR tomorrow.

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-sim/pull/2573

Waiting for your answer for "If I understand well, "Known issue of command line tools" is not needed anymore because of this solved issue https://github.com/gazebosim/gz-tools/issues/8 . If yes, I can also remove it."

azeey, 2024-09-05:
Do you mind creating similar PR as https://github.com/gazebosim/gz-launch/pull/275 to fix the "known issue" section?

avanmalleghem, 2024-09-06:
Here it is : https://github.com/gazebosim/gz-sim/pull/2603

azeey, 2024-09-06:
Thanks for testing and the PRs!!

closed 2024-09-06 by azeey