gazebosim/gazebo_test_cases #1122: gz-sim: Adding system plugins

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/adding_system_plugins.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/adding_system_plugins.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-06:
- Update link [This Gazebo tutorial](https://gazebosim.org/api/sim/8/createsystemplugins.html)
- gz-sim/gz-sim8/ to gz-sim9
- typo "appling force"

But the tutorial works as expected, that's a pretty good news !!
Will create a PR

avanmalleghem, 2024-09-06:
Here it is : https://github.com/gazebosim/gz-sim/pull/2602

ahcorde, 2024-09-09:
PR is merged, thank you for testing!

closed 2024-09-09 by ahcorde