gazebosim/gazebo_test_cases #1127: gz-sim: Apply Force and Torque

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/sim/9/apply_force_torque.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/apply_force_torque.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Yadunund, 2024-09-02:
## Setup
* M1 Macbook Pro 14.6.1
* Ionic built from source

## Apply force to a link

[image]
Successfully tested the tutorial. 

## Apply torque to a link

[image]

## Apply force with an offset

[image]

## Rotation tool

[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde