gazebosim/gazebo_test_cases #1137: gz-sim: Blender Python script for programmatic mesh distortion

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/blender_distort_meshes.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/blender_distort_meshes.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-05:
## Ubuntu 24.04 Noble - Blender 3.2

**From shell command prompt**
```bash
combuster@combuster54:~/Desktop$ blender -b -P distort_mesh.py -- '/home/combuster/Desktop/models/Coke/meshes/coke.obj' 'LPCoke_Cube' 0.005 "['subdiv_mod', 'vert_rand', 'edge_subdiv']"
Blender 3.2.0 (hash e05e1e369187 built 2022-06-08 10:37:43)
(  0.0000 sec |   0.0000 sec) Importing OBJ '/home/combuster/Desktop/models/Coke/meshes/coke.obj'...
  (  0.0001 sec |   0.0000 sec) Parsing OBJ file...
    (  0.0024 sec |   0.0023 sec) Done, loading materials and images...
    (  0.0041 sec |   0.0040 sec) Done, building geometries (verts:190 faces:356 materials: 2 smoothgroups:1) ...
    (  0.0074 sec |   0.0073 sec) Done.
  (  0.0074 sec |   0.0074 sec) Finished importing: '/home/combuster/Desktop/models/Coke/meshes/coke.obj'
Progress: 100.00%

Distorting mesh [LPCoke_Cube.000] for relative extent 0.005 out of [0, 1]...
Subdivision modifier level is 0, skipping
Applying mesh vertex randomization with offset 0.007235950231552124...
{'FINISHED'}
Applying edge subdivide with number of cuts 1...
{'FINISHED'}
    (  0.0001 sec |   0.0000 sec) OBJ Export path: '/home/combuster/Desktop/models/Coke/meshes/coke_distort.obj'
          (  0.0335 sec |   0.0333 sec) Finished writing geometry of 'LPCoke_Cube.000'.
      (  0.0337 sec |   0.0336 sec) Finished exporting geometry, now exporting materials
      (  0.0341 sec |   0.0340 sec) OBJ Export Finished
Progress: 100.00%

Exported result to [/home/combuster/Desktop/models/Coke/meshes/coke_distort.obj]

Blender quit
```
[image]

**From Blender Scripting panel Python prompt**

[image]

[image]

Combuster54, 2024-09-06:
**To add a new distortion method to the script**

```python
#!/usr/bin/env python3

# In Blender, perform mesh modifications.
#
# Usage:
#   This script is to be run from within the Blender GUI. Tested in Blender
#   2.92, 3.0.1 and 3.2.
#   distort_mesh.py <mesh_path> <object_prefix> [distort_extent] [method]
#   where
#   mesh_path: Absolute path to mesh file
#   object_prefix: Prefix of object name in Scene Collection in Blender. The
#     prefix instead of the name is required, because if the script is
#     repeatedly run, Blender will append numerical suffixes .001, .002, etc.
#     to the object name.
#   distort_extent: A floating point number in the range of [0, 1]
#   method: A list of strings. Distortion operations, in order of desired
#     execution.
#
#   Example from command line:
#   $ blender -b -P distort_mesh.py -- /path/to/mesh.obj object_prefix 0.005 "['subdiv_mod', 'vert_rand', 'edge_subdiv']"
#
#   Example from Blender Scripting GUI Python command line:
#   >>> file_path = '/path/to/mesh.dae'
#   >>> object_prefix = 'Cube'
#   >>> distort_extent = 0.005  # float in range [0, 1]
#   >>> method = ['subdiv_mod', 'vert_rand', 'edge_subdiv']
#   >>> sys.argv = [file_path, object_prefix, distort_extent, method]
#   >>> exec(open('/path/to/distort_mesh.py').read());
#

import bpy

import os
import sys

import numpy as np

def find_target_object(object_prefix):

    target_obj = None
    object_name = None

    # Find the object that matches the name prefix. Assume the first one.
    for obj in bpy.data.objects:
        if obj.name.startswith(object_prefix):
            target_obj = bpy.data.objects[obj.name]
            object_name = obj.name
            break
    if target_obj == None:
        print('ERROR: Object with prefix [{}] not found'.format(object_prefix))
        return None

    return target_obj, object_name

# shading: 'WIREFRAME', 'SOLID', 'MATERIAL', or 'RENDERED'
def viewport_shading(shading):
    print('Changing viewport shading to {}...'.format(shading))
    areas = bpy.context.workspace.screens[0].areas
    for ar in areas:
        for spc in ar.spaces:
            if spc.type == 'VIEW_3D':
                spc.shading.type = shading

def subdivision_modifier(obj, levels=2):

    if levels == 0:
        print('Subdivision modifier level is 0, skipping')
[... 182 lines cut ...]
            argStartI = i + 1
            break

    # Parse args
    if len(sys.argv) - argStartI < 2:
        print('ERROR: Mesh name prefix not provided. No object to distort.')
    else:
        file_path = sys.argv[argStartI]
        object_prefix = sys.argv[argStartI + 1]
        if len(sys.argv) - argStartI > 1:
            distort_extent = float(sys.argv[argStartI + 2])
        if len(sys.argv) - argStartI > 2:
            # If arg is not already a list (happens when script run from
            # bash command line, for example, as opposed to Blender Python
            # prompt), convert string literal of a list, to a list.
            if type(sys.argv[argStartI + 3]) != list:
                import ast
                method = ast.literal_eval(sys.argv[argStartI + 3])

        distort(file_path, object_prefix, distort_extent, method)
```

[image]

[image]

jennuine, 2024-09-06:
Thanks @Combuster54 !

closed 2024-09-06 by jennuine