gazebosim/gazebo_test_cases #1143: gz-sim: Blender SDF Exporter

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/blender_sdf_exporter.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/blender_sdf_exporter.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Combuster54, 2024-09-05:
Interested

Combuster54, 2024-09-05:
## Ubuntu 24.04 Noble

**Using the Blender SDF Exporter**

[image]

**default main**
[image]

**alternative main**
[image]

both works as expected.

closed 2024-09-05 by azeey