gazebosim/gazebo_test_cases #1155: gz-sim: Create System Plugins

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/createsystemplugins.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/create_system_plugins.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-06:
I'm gonna open a PR to :
- Update the "Setup the build" code
- Add a link to the fully implemented example : https://github.com/gazebosim/gz-sim/blob/gz-sim9/examples/plugin/system_plugin

avanmalleghem, 2024-09-06:
Here we go : https://github.com/gazebosim/gz-sim/pull/2601

ahcorde, 2024-09-09:
PR is merged, thank you for testing!

closed 2024-09-09 by ahcorde