gazebosim/gazebo_test_cases #1158: gz-sim: Create a maritime vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/create_vehicle.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/create_vehicle.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alexandrosnic, 2024-09-03:
Interested tackling this issue

alexandrosnic, 2024-09-03:
Replaced single with double quotes in xml, and fixed the path to the image in the Source link.
[Opened a PR](https://github.com/gazebosim/gz-sim/pull/2583)

iche033, 2024-09-04:
https://github.com/gazebosim/gz-sim/pull/2592

closed 2024-09-04 by iche033