gazebosim/gazebo_test_cases #1164: gz-sim: Detachable Joints

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/detachablejoints.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/detachable_joints.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 15-10-19.webm](https://github.com/user-attachments/assets/b8772d7b-7dce-4fce-8006-37a67b53076c)
A lot of interesting issues here, see video

david-dorf, 2024-08-31:
Has warnings pop up when loading the world, and crashes upon using "attach" publisher

avanmalleghem, 2024-09-01:
@david-dorf please avoid doing it before saying you are interested in... So that we don't do the job twice... Follow the process described in https://community.gazebosim.org/t/gazebo-ionic-test-and-tutorial-party-instructions/2989

azeey, 2024-09-03:
@avanmalleghem, if you still want to test this tutorial, you can get credit for it. Otherwise, I'll assign it to @david-dorf

azeey, 2024-09-04:
> I'm more here to contribute to the community than to get credit so will do another one ! Thanks for the proposition @azeey

Awesome!! ❤️

closed 2025-08-27 by azeey