gazebosim/gazebo_test_cases #1173: gz-sim: ERB Template

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/erbtemplate.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/erb_template.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-05:
[image]
[image]

Nothing to add ! This one can be closed aswell

ahcorde, 2024-09-05:
thank you for testing!

closed 2024-09-05 by ahcorde