gazebosim/gazebo_test_cases #1176: gz-sim: Frame of reference

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/frame_reference.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/frame_reference.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-09-03:
I'm interested , can I get more info about what to test in this as nothing is mention properly in given links

azeey, 2024-09-03:
For this, just checking for any typos or factual errors seems sufficient.

akky20, 2024-09-05:
## There are some typos 

# 1
[image]
Original: "differente choices"
Correction: "different choices"

# 2
[image]
uses "z" in lowercase, while earlier in the document "Z" was capitalized. For consistency, it might be better to use uppercase "Z" here as well.

# 3
[image]
Original: "desigining your mesh"
Correction: "designing your mesh"

ahcorde, 2024-09-05:
@akky20 do you mind to open a PR ?

akky20, 2024-09-05:
@ahcorde sure will do

akky20, 2024-09-05:
> @akky20 do you mind to open a PR ?

I have added a PR for this , please have a look

closed 2025-08-27 by azeey