gazebosim/gazebo_test_cases #1182: gz-sim: Headless Rendering

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/headless_rendering.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/headless_rendering.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

j-rivero, 2024-08-29:
we might need a way to check that the EGL is actually working.

Tested using docker on i915 Intel with access to the video group. The run dees not seem to have any problem but I can not tell if it is doing what is suppose to do following the tutorial.

azeey, 2024-09-03:
One way to check is to run it after unsetting the `DISPLAY` environment variable: `DISPLAY= gz sim -s --headless-rendering ...` and on another terminal run `gz sim -g` and check that the camera sensor is working

iche033, 2024-09-04:
the ogre2.log should also say if we're using EGL:

```sh
gz sim -v 4 -s --headless-rendering sensors_demo.sdf

...

grep -R "EGL Subsystem" ~/.gz/rendering/ogre2.log
```

output should not be empty:

```
*** Starting EGL Subsystem ***
*** Starting EGL Subsystem ***
*** Stopping EGL Subsystem ***
```

peci1, 2024-09-09:
I've tested this locally. There are a few improvements for the tutorial: https://github.com/gazebosim/gz-sim/pull/2607 .

I did the tests by converting the Docker image to a Singularity/Apptainer image and running locally. I could not test the AWS part as I've no account there.

j-rivero, 2024-09-09:
Thanks @peci1 merged.

closed 2024-09-09 by j-rivero