gazebosim/gazebo_test_cases #1185: gz-sim: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/install.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
## Ubuntu 24.04 (Noble)

### Binary Install
Everything works fine

### Source Install

j-rivero, 2024-09-02:
> Two test case failed as follow:

I think that we did not manage to get all the tests to pass on gz-sim in a consistent manner. Instructions seems good, thanks for testing.

closed 2024-09-02 by j-rivero