gazebosim/gazebo_test_cases #1186: gz-sim: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/install.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-11:
I'll test this over the weekend on a real Win 10 laptop (if it's possible to install gz-sim on Windows).

DanielFNG, 2024-09-11:
(Sorry to jump in on this issue, but I wanted to install Gazebo on Windows to help with the issues and immediately ran in to some problems trying to follow these instructions.)

The links above don't seem to give any instructions for installing Windows?

There are Windows-specific instructions at this link: https://gazebosim.org/docs/ionic/install_windows/. However, the latest versions of Gazebo available for me via conda are gz-sim8. The results of a `conda search` are:

```
gz-sim8                        8.3.0      h57928b3_0  conda-forge
gz-sim8-python                 8.3.0  py39ha4551ba_0  conda-forge
```

peci1, 2024-09-16:
I've added the install instructions in: https://github.com/gazebosim/gz-sim/pull/2628 .

closed 2024-09-16 by azeey