gazebosim/gazebo_test_cases #1188: gz-sim: Joint Controllers

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/jointcontrollers.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/joint_controller.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

bperseghetti, 2024-09-13:
So a few things I guess, first is that the documentation is slightly broken here is a PR to fix it. The only other thing is that it does not report the progress of the joint trajectory controller.

https://github.com/user-attachments/assets/50b50d8c-1ba0-4ab5-8b83-1739874b4779

Error with trying to use example2.sdf without XML fix:
[image]

ahcorde, 2024-09-16:
PR is merged, Thank you for testing!

closed 2024-09-16 by ahcorde

bperseghetti, 2024-09-16:
Been a crazy few weeks for me, I forgot to somehow push my last commit up to fix the joint trajectory portion of tutorial: https://github.com/gazebosim/gz-sim/pull/2629

My bad on that.