gazebosim/gazebo_test_cases #1191: gz-sim: Create a lander vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/lander.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/lander.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-09-01:
Interested in !

avanmalleghem, 2024-09-01:
- Path are from gz-sim8, must be adapted
- this path "gz sim ~/my_models/my_lander/model.sdf" is wrong
- Second screenshot is incorrect, we should not see the black box in it

Will create a PR very soon ! This tutorial was really fun.

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-sim/pull/2576

ahcorde, 2024-09-03:
Thank you for testing!

closed 2024-09-03 by ahcorde