gazebosim/gazebo_test_cases #1197: gz-sim: Light config

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/light_config.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/light_config.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

akky20, 2024-09-03:
## Works fine 

[Screencast from 04-09-24 03:04:38 AM IST.webm](https://github.com/user-attachments/assets/d77e8401-7a7d-4e6b-8d4f-1616aab231f2)

kindly ignore the terminal on left side

azeey, 2024-09-03:
Thanks for testing!

closed 2024-09-03 by azeey