gazebosim/gazebo_test_cases #1203: gz-sim: Logical Audio Sensor

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/logicalaudiosensor.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/logical_audio_sensor.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

csimon-be, 2024-08-30:
Interested in this one

akky20, 2024-09-05:
If you are not interested in tesing this , as its been a week, can I do inplace of you ?

csimon-be, 2024-09-05:
Hi @akky20,
I have been delayed, feel free to test it and to communicate your results. I will focus on the others assigned to me first.

akky20, 2024-09-05:
thankyou csimon-be  ,

## Here  the topics
[image]

## No output with this topic as expected
[image]

## got output for second topic
[image]

## Here are the services 
[image]

[image]

[image]

closed 2024-09-05 by azeey