gazebosim/gazebo_test_cases #1209: gz-sim: Case study: migrating the ArduPilot ModelPlugin from Gazebo classic to Gazebo

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/ardupilot.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/migrating_ardupilot_plugin.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-09-05:
I'm interested , can I get more info about what to test in this , like do I need to find any typo or any factual error ?

iche033, 2024-09-05:
yeah, check links, typos, formatting etc. The tutorial points to this repo: https://github.com/gerkey/ardupilot_gazebo/tree/ignition. I just noticed that the code in there is still using ignition and likely won't compile in Ionic. We'll need to add a note in the tutorial to mention this.

closed 2025-08-27 by azeey