gazebosim/gazebo_test_cases #1212: gz-sim: Migration from Gazebo-classic: Actor API

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/migrationactorapi.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/migration_actor_api.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-09-05:
I'm interested , can I get more info about what to test in this .

iche033, 2024-09-05:
you can check links, typos, formatting, etc.  For reference, this is the [Actor](https://github.com/gazebosim/gz-sim/blob/gz-sim9/include/gz/sim/Actor.hh) class. You can also verify that the APIs mentioned in the tutorial has not changed and consistent with the ones in the source code.

closed 2025-08-27 by azeey