gazebosim/gazebo_test_cases #1230: gz-sim: Migration from Gazebo classic: SDF

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/migrationsdf.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/migration_sdf.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

EnzoGhisoni, 2024-08-30:
I'm interested to test it

EnzoGhisoni, 2024-08-31:
Step 1 - Instructions work
The tutorial is working as expected (video attached in the step2). The only issues is that we have error messages (that are not preventing the tutorial to work) in the simulation terminal:

(2024-09-01 00:38:28.688) [error] Failed to load system plugin [libgazebo_ros_control.so] : Could not find shared library.
(2024-09-01 00:38:28.689) [error] Failed to load system plugin [libgazebo_ros_camera.so] : Could not find shared library.
(2024-09-01 00:38:28.689) [error] Failed to load system plugin [libgazebo_ros_gpu_laser.so] : Could not find shared library.
(2024-09-01 00:38:28.786) [error] [GUI] Unable to find file with URI [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Could not resolve file [model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Unable to find file[model://rrbot_description/meshes/hokuyo.dae]
(2024-09-01 00:38:28.786) [error] [GUI] Failed to load geometry for visual: link3_fixed_joint_lump__hokuyo_link_visual_2

I think it's mostly related to the fact that the service has issues to find the dependante file of the URDF. It could be interesting to indicate a bit more the setup required to make work the URDF or provide an URDF without meshes file imported to test it in standalone.

Step 2 - Images (if there are any) match the result
[record_gazebo_sim.webm](https://github.com/user-attachments/assets/28818b10-f04d-47ea-b428-7fc176ff49e7)

EnzoGhisoni, 2024-09-01:
Step 1 - Instructions work
The SDF example file is working (image in step2). The links to [ERB](erb_template.html),  [Server configuration tutorial](server_config.html) and [GUI configuration tutorial](gui_config.html) are dead.

I will change the links of ERB page by these ones:
https://github.com/gazebosim/gz-sim/blob/gz-sim9/tutorials/erb_template.md
https://github.com/gazebosim/gz-sim/blob/gz-sim9/tutorials/server_config.md
https://github.com/gazebosim/gz-sim/blob/gz-sim9/tutorials/gui_config.md

The rest of the tutorial is working for me.
Step 2 - Images (if there are any) match the result

[image]

ahcorde, 2024-09-03:
Thank you for testing and the PR!

closed 2024-09-03 by ahcorde