gazebosim/gazebo_test_cases #1239: gz-sim: Model and optimize meshes in Blender for simulations

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/model_and_optimize_meshes.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/model_and_optimize_meshes.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Combuster54, 2024-09-05:
**Make a cylinder**

[image]

**Move Gizmo**

[image]

**Rotate Gizmo**

[image]

**Scale Gizmo**

[image]

**Bevel**

[image]

**Extrude**

[image]

**Loop cut**

[image]

**Duplicate**

Add the command `Shift D`

[image]

**Exporting models**

[image]

**Troubleshooting**

Check Scale/Location
[image]

Check Pivot Point/Shape Origins

[image]

ahcorde, 2024-09-05:
Thank you for testing!

closed 2024-09-05 by ahcorde