gazebosim/gazebo_test_cases #1245: gz-sim: Model Photo Shoot

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/model_photo_shoot.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/model_photo_shoot.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alesof, 2024-08-31:
Tutorial is clear and instructions work. 

Although tutorial contains values that differ from the default sdf. [Opened a PR](https://github.com/gazebosim/gz-sim/pull/2562) that contains the updates.

This is also related to #1650.

ahcorde, 2024-09-03:
Thank you for testing!