gazebosim/gazebo_test_cases #1251: gz-sim: Move Camera to model

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/move_camera_to_model.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/move_camera_to_model.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-30:
Interested in !

avanmalleghem, 2024-08-30:
Works like a charm ! Maybe we should add that this tutorial is on the "shapes.sdf" world ? Otherwise, can be closed.

iche033, 2024-08-30:
> Maybe we should add that this tutorial is on the "shapes.sdf" world ?

yeah that'll be nice. Can you open a pull request for this?

avanmalleghem, 2024-09-02:
PR opened : https://github.com/gazebosim/gz-sim/pull/2571

ahcorde, 2024-09-02:
PR is merged, thank you for testing!

closed 2024-09-02 by ahcorde