gazebosim/gazebo_test_cases #1257: gz-sim: Pause and Run simulation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/pause_run_simulation.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/pause_run_simulation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

csimon-be, 2024-08-30:
Interested in this one

Combuster54, 2024-09-09:
## Ubuntu 24.04 Noble

**Transport API**
 start and pause simulation

[image]

**GUI**

[image]

✅  Works as expected

csimon-be, 2024-09-10:
Working fine, it can be closed

ahcorde, 2024-09-10:
Thank you for testing!