gazebosim/gazebo_test_cases #1258: gz-sim: Pause and Run simulation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/pause_run_simulation.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/pause_run_simulation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-11:
I'll test this over the weekend on a real Win 10 laptop.

Combuster54, 2024-09-11:
## Windows 10

**Transport API**

[image]

**GUI**
[image]

Works as expected

closed 2024-09-11 by azeey