gazebosim/gazebo_test_cases #1260: gz-sim: Physics engines

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/physics.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/physics.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-02:
Interested

Combuster54, 2024-09-03:
## Ubuntu 24.04 Noble

**Setting helloWorldPlugin**
```bash
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ sudo ln -s ~/test/gazebo-ionic-physics/hello_world_loader/build/libHelloWorldPlugin.so /usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ sudo ln -s ~/test/gazebo-ionic-physics/hello_world_loader/build/libHelloWorldPlugin.so /home/physics_engines/libHelloWorldPlugin.so
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_plugin/build$ export GZ_SIM_PHYSICS_ENGINE_PATH=/home/physics_engines
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_plugin/build$ echo $GZ_SIM_PHYSICS_ENGINE_PATH
/home/physics_engines
```

**Tell Gazebo what engine to load**

**From SDF**

```xml
<?xml version="1.0" ?>
<sdf version="1.7">
  <world name="custom_world">

    <plugin
      filename="libHelloWorldPlugin.so"
      name="mock::HelloWorldPlugin">
      <engine>
        <filename>HelloWorld</filename>
      </engine>
    </plugin>

    <light name="sun" type="directional">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 -0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.5 -1</direction>
    </light>

  </world>
</sdf>
```

```bash
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ ./hello_world_loader ~/test/gazebo-ionic-physics/hello_world_plugin/build/libHelloWorldPlugin.so 
Testing plugin: mock::HelloWorldPlugin
  engine name: HelloWorld
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ ./hello_world_loader /home/physics_engines/libHelloWorldPlugin.so 
Testing plugin: mock::HelloWorldPlugin
  engine name: HelloWorld
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ ./hello_world_loader /usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so
Testing plugin: mock::HelloWorldPlugin
  engine name: HelloWorld
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ ls /home/physics_engines/libHelloWorldPlugin.so /usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so
/home/physics_engines/libHelloWorldPlugin.so
/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so
combuster@combuster54:~/test/gazebo-ionic-physics/hello_world_loader/build$ cd
combuster@combuster54:/home$ gz sim ~/test/gazebo-ionic-physics/world/physic_plugin_world.sdf
(2024-09-03 14:22:37.770) [error] Failed to load system plugin [libHelloWorldPlugin.so] : Could not find shared library.
```

**error**
`(2024-09-03 14:22:37.770) [error] Failed to load system plugin [libHelloWorldPlugin.so] : Could not find shared library.`

**From command**
Trying with other names... same output

```bash
combuster@combuster54:~$ gz sim --physics-engine HelloWorld
(2024-09-03 14:13:02.207) [error] Failed to find plugin [HelloWorld]. Have you checked the GZ_SIM_PHYSICS_ENGINE_PATH environment variable?
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
^Ccombuster@combuster54:~$ gz sim --physics-engine libHelloWorldPlugin
(2024-09-03 14:13:46.156) [error] No physics plugins found in library [/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libHelloWorldPlugin.so].
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
^Ccombuster@combuster54:~$ 
```

Using absolute path inside filename

```xml
  filename="/home/physics_engines/libHelloWorldPlugin.so"
```

```bash
combuster@combuster54:/home$ gz sim ~/test/gazebo-ionic-physics/world/physic_plugin_world.sdf
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
```
works as expected.

I'm not sure if I've done something wrong... any feedback?

I create this repo to make these tests:
https://github.com/Combuster54/gazebo-ionic-physics

iche033, 2024-09-06:
Here's the updated SDF file that should fix the issue:

```xml
<?xml version="1.0" ?>
<sdf version="1.7">
  <world name="custom_world">

    <plugin
      filename="gz-sim-physics-system"
      name="gz::sim::systems::Physics">
      <engine>
        <filename>HelloWorld</filename>
      </engine>
    </plugin>

    <light name="sun" type="directional">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 -0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.5 -1</direction>
    </light>

  </world>
</sdf>
```

This may be a little confusing but to clarify, there are different types of plugins. Your original SDF file tells gz sim to find and load a gz-sim system (also known as a plugin) named HelloWorldPlugin.

In the case of this tutorial, we want to load a "gz-physics plugin". This type of plugin is loaded by the gz-sim-physics-system. Effectively what gz sim is doing is that it first loads a gz-sim-physics-system which in turn loads a gz-physics plugin. So we'll need to keep the `gz-sim-physics-system` plugin in the SDF file and tell it to load the `HelloWorld` gz physics plugin. Hopefully that clears things up.

Combuster54, 2024-09-06:
Oh... I try with this sintax before... I thought that I was wrong so I decided to find another way. 

[image]

So this is correct?

Combuster54, 2024-09-06:
Perhaps we can add a note inside this tutorial related to 
`(2024-09-06 00:50:05.722) [error] [Physics.cc:854] No physics plugins found in library [/home/iche/code/gz_i_ws_noble2/src/gz-physics/examples/hello_world_plugin/build/libHelloWorldPlugin.so].`

iche033, 2024-09-06:
updated the error msg and added an entry in the troubleshooting section about this msg.

closed 2024-09-06 by iche033