gazebosim/gazebo_test_cases #1264: gz-sim: Converting a Point Cloud to a 3D Model

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/pointcloud.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/point_cloud_to_mesh.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-09:
I'll test this.

peci1, 2024-09-16:
The tutorial works.

However, the recent Linux version changes broke the rendering of https://gazebosim.org/api/sim/9/pointcloud.html . There are 2 issues:

1. All images reference main branch. I think they should reference gz-sim9.
2. The `Error "Visibility table not instantiated"` image doesn't render at all as image - it gets instead its HTML tag printed in the content. I'm not sure it it's the double quotes in the image description or if there is some other thing the renderer doesn't like.