gazebosim/gazebo_test_cases #1284: gz-sim: Spherical coordinates

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/spherical_coordinates.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/spherical_coordinates.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-31:
interested

akky20, 2024-09-03:
## Spherical Coordinate SDF with Service list  
[image]

## Moon SDF with service list
[image]

azeey, 2024-09-03:
@akky20 Did calling the services per the tutorial work?

akky20, 2024-09-03:
@azeey yes its working correclty 
[Screencast from 04-09-24 03:51:36 AM IST.webm](https://github.com/user-attachments/assets/76e51a24-4ba1-4c49-a1e5-d7339f1f4a66)

azeey, 2024-09-03:
Thanks for confirming!

closed 2024-09-03 by azeey