gazebosim/gazebo_test_cases #1285: gz-sim: Spherical coordinates

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/spherical_coordinates.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/spherical_coordinates.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-16:
This tutorial mostly works. But there are differences on Windows regarding handling of multiline commands that can't just be copied from the tutorial page into Windows console. This is the best I got to work:

```
gz service -s /world/spherical_coordinates/create ^
--reqtype gz.msgs.EntityFactory ^
--reptype gz.msgs.Boolean ^
--timeout 300 ^
--req 'sdf: "<?xml version=\"1.0\" ?><sdf version=\"1.6\"><model name=\"spawned_model\"><link name=\"link\"><visual name=\"visual\"><geometry><sphere><radius>1.0</radius></sphere></geometry></visual><collision name=\"visual\"><geometry><sphere><radius>1.0</radius></sphere></geometry></collision></link></model></sdf>" ^
spherical_coordinates: {latitude_deg: 35.59999, longitude_deg: 140.09999, elevation: 11.0}'
```

peci1, 2024-09-16:
There's also a segfault with dem_moon.sdf: https://github.com/gazebosim/gz-sim/issues/2626 .

azeey, 2024-09-16:
We have a lot of multiline commands in our tutorials, so going through and updating them to have windows alternatives would not be feasible.  It also depends on whether you're using CMD or PowerShell, so I'm inclined to leave it as is and hope that users can replace the appropriate delimiter.

closed 2024-09-16 by azeey