gazebosim/gazebo_test_cases #1288: gz-sim: Create a surface vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sim/9/surface_vehicles.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/surface_vehicles.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

knmcguire, 2024-09-13:
Alright, this will conclude my Maritime adventure 😄 ⛵  Sign me up for this one for windows 11 !

knmcguire, 2024-09-13:
There are issues with this tutorial ❌ 

I had to implement some fixes where I could but at one point I couldn't continue.
1. The link `wget https://raw.githubusercontent.com/gazebosim/gz-sim/gz-sim8/tutorials/files/surface_vehicles/gz_maritime_ws.zip -O ~/gz_maritime_ws.zip` points to gz-sim8, so I changed it to gz-sim9
2. Even with the change to gz-sim9, the cmakefile was still tailored towards gz-sim8 in terms of dependencies.
3. I got an `M_PI flagged as undeclared identifier` error so I had to add '#define _USE_MATH_DEFINES #include <cmath>' to Wavefield.cc
4. After this I could still not pass colcon build since I got a linker error `LINK : fatal error LNK1181: cannot open input file 'Release\Waves.lib'`

That's where I hit my limit in debugging, so Unfortunately it seems that this tutorial needs to be updated by someone more experienced with this.

j-rivero, 2024-09-13:
Thanks @knmcguire for testing it and the hints. I follow the debugging. The problem with the linking can be solve by either use the black magic of CMake in the form on enabling a variable:
```
set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)
```
or just change libraries to be static and remove the need of .lib file:

``` 
# Waves
add_library(Waves STATIC
  src/Wavefield.cc)
```

More hints:
 * jsoncpp was not installed on conda
 * The `M_PI` code can be just replaced with `GZ_PI` (in gazebo-common)
 * All the references to Ionic need to be replaces in the CMakeLists.txt
 * The world file needs to remove the reference to lib in the plugin name

Almost there:
[image]

Something is happening with the loading of the png and materials since the files are in my system.

I'll prepare the PR.

knmcguire, 2024-09-14:
Thanks! I can try again once that is working again:) I'll be going on a holiday so I won't be able to until after September 22nd