gazebosim/gazebo_test_cases #1296: gz-sim: Buoyancy

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/theory_buoyancy.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/theory_buoyancy.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-02:
Interested

Combuster54, 2024-09-03:
## Ubuntu 24.04 Noble

`The first submarine is neutrally buoyant, the second sinks, and the third floats. Run the following command to see an example of a buoyancy plugin configured with a uniform fluid density`

**The buoyancy plugin**
[image]

**Graded buoyancy**
`You should be able to see a sphere bobbing up and down undergoing simple harmonic motion on the surface of the fluid (this is expected behavior as the harmonic motion is usually damped by the hydrodynamic forces.`

[image]

Everything seems fine

closed 2024-09-03 by azeey