gazebosim/gazebo_test_cases #1299: gz-sim: Hydrodynamics

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/theory_hydrodynamics.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/theory_hydrodynamics.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-09-02:
Interested

Combuster54, 2024-09-03:
## Ubuntu 24.04 Noble

**A Simple example**
[image]

**first change**

```xml
<xUabsU>-0.32282</xUabsU>
<xU>-2.5</xU>
```
[image]
**second change**

```xml
<xUabsU>-3.5</xUabsU>
<xU>-20</xU>
```
[image]

azeey, 2024-09-05:
@arjo129 I know the error message about added mass has been changed to a warning in https://github.com/gazebosim/gz-sim/pull/2579, but would it be possible to change the parameters in our examples so we don't have the warnings?

Combuster54, 2024-09-06:
@azeey Yes. There is a way to do that.

**Solution**

Add
```xml
        <disable_added_mass>true</disable_added_mass>
```

```xml
      <!-- Uncomment to add hydrodynamics -->
      <plugin
        filename="gz-sim-hydrodynamics-system"
        name="gz::sim::systems::Hydrodynamics">
        <link_name>base_link</link_name>
        <disable_added_mass>true</disable_added_mass>
        <xDotU>-0.04876161</xDotU>
        <yDotV>-1.26324739</yDotV>
        <zDotW>-1.26324739</zDotW>
        <kDotP>0</kDotP>
        <mDotQ>-0.3346</mDotQ>
        <nDotR>-0.3346</nDotR>
        <xUabsU>-0.62282</xUabsU>
        <xU>-5</xU>
        <yVabsV>-60.127</yVabsV>
        <yV>-5</yV>
        <zWabsW>-6.0127</zWabsW>
        <zW>-100</zW>
        <kPabsP>-0.001916</kPabsP>
        <kP>-1</kP>
        <mQabsQ>-6.32698957</mQabsQ>
        <mQ>-1</mQ>
        <nRabsR>-6.32698957</nRabsR>
        <nR>-1</nR>
      </plugin>
```

[image]