gazebosim/gazebo_test_cases #1302: gz-sim: Triggered Publisher

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/triggeredpublisher.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/triggered_publisher.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-31:
interested

akky20, 2024-09-03:
## Working fine

[Screencast from 03-09-24 08:42:41 PM IST.webm](https://github.com/user-attachments/assets/0714b330-f18c-419a-874d-f2f29672a6da)

azeey, 2024-09-03:
@jc-sz you said you were interested 3 days ago, but there hasn't been any updates, so I'm going to credit this to @akky20