gazebosim/gazebo_test_cases #1305: gz-sim: Create an underwater vehicle

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sim/9/underwater_vehicles.html)
* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/tutorials/underwater_vehicles.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-31:
Interested

Combuster54, 2024-08-31:
## Ubuntu 24.04 Noble

**Create your vehicle**

[image]

**Adding buoyancy**
[image]

**Adding thrusters**

[image]

**Adding hydrodynamics**

[image]
Uncomment hydrodynamics plugin shows this error. Despite this, the robot moves correctly. 

```bash
(2024-08-31 14:49:02.472) [error] The use of added mass through this plugin is deprecated and willbe removed in Gazebo J* as this formulation has instabilities. We recommend using the SDF `<fluid_added_mass>` tag based method[http://sdformat.org/spec?ver=1.11&elem=link#inertial_fluid_added_mass]To get rid of this warning we recommend setting`<disable_added_mass> to true.
```

**Adding control surfaces**

[image]

stop underwater vehicle

[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde